Legion2/Somfy_Remote_Lib #1: Pinout schematic

tshabarek, 2021-02-17:
Hi all
Can Someone please help me with the wiring schematic
Arduino uno with cc101
Thanks

Legion2, 2021-02-17:
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib#wiring

tshabarek, 2021-02-17:
Thank you legion2
I will try these wiring with my somfy device and let you know

tshabarek, 2021-02-23:
Dear legion2
I tried to connect CC1101 and Arduino Uno 
according to Lsatan library :https://github.com/LSatan/SmartRC-CC1101-Driver-Lib#wiring
When I press in serial monitor any number from 1 to 8 nothing is happening I even tried to write the name of the orders ,I think it should send me back this message  "finished sending"
	#ifdef DEBUG
		Serial.println("finished sending");

but nothing is happening ,what did I do wrong?
Thank you very much

Legion2, 2021-02-23:
> should send me back this message "finished sending"

only if you put `#define DEBUG` in the top of the sketch

tshabarek, 2021-02-23:
Yes right when I added #define DEBUG it starts to send message.

But the smofy motor didn't responce to my attempts
I pressed the Prog button in the original somfy motor and the motor moved up and down waiting for the new remote
The I pressed number 8 in the serial monitor but the somfy device didnt responce.
I wonder if I should use logic level converter in order to work?
Thank you

Legion2, 2021-02-23:
did you setup the CC1101 correncty and did you use https://github.com/Legion2/Somfy_Remote_Lib/blob/main/examples/CC1101/CC1101.ino?

tshabarek, 2021-02-23:
Yes I used "CC1101.ino"  file and uploaded it to the arduino device and I did the wirinig as shown
I trued Arduino mega and Uno
Is there anytjing should be setted up in the code?

Legion2, 2021-02-23:
Can you enable the `DEBUG` in every file, so we get more serial output? Either you add `#define DEBUG` to all the library files or you add `uno.build.extra_flags=-DDEBUG` to `AppData\Local\arduino15\packages\arduino\hardware\avr\1.8.3\boards.txt` below the other `uno.` lines.

tshabarek, 2021-02-23:
Okay I will apply debbuging to all files
I will try Esp32 with it tommorrow and let you know.
Thanks

tshabarek, 2021-03-23:
Hi  Leon,
I tried arduini mega with the cheap RF module after changing the crystals to 433.42mhz 
It is working well.
But I couldn't know how to change between channels when using multi channel code.
Can you please help me with.
Your work most appreciated.
Thank you,

Legion2, 2021-03-23:
`MultipleRemotes.ino` is only an example to demonstrate how it works, you have to modify it for your use case and implement the selection of the remote yourself.

closed 2021-03-31 by Legion2