Legion2/Somfy_Remote_Lib #2: How to change chanel of the multi remote example

thommio, 2021-04-13:
Hi,

I couldn't find how to change channel on the multi remote example. Could you explain?

Got your single remote code working on my esp8266 using the Blynk app on IOS and Android, so thanks :)

Legion2, 2021-04-13:
In the mutli remote example, the commands are send to all "channels" at the same time. It is only for demonstration, if you want to select a specific "channel" (SomfyRemote) you have to write some code, which only sends the command with one of the many remotes.

thommio, 2021-04-13:
Aah i see now! thanks.

I made it like this now, should work.

`#define BLYNK_PRINT Serial
#define EMITTER_GPIO 2
#define EEPROM_ADDRESS 0
#define REMOTE1 0x5184c8
#define REMOTE2 0x25b5d5
#define REMOTE3 0xc6c78f
#define REMOTE4 0x59714b

#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <EEPROM.h>
#include <EEPROMRollingCodeStorage.h>
#include <SomfyRemote.h>

EEPROMRollingCodeStorage rollingCodeStorage1(0);
EEPROMRollingCodeStorage rollingCodeStorage2(2);
EEPROMRollingCodeStorage rollingCodeStorage3(4);
EEPROMRollingCodeStorage rollingCodeStorage4(6);
SomfyRemote somfyRemote1(EMITTER_GPIO, REMOTE1, &rollingCodeStorage1);
SomfyRemote somfyRemote2(EMITTER_GPIO, REMOTE2, &rollingCodeStorage2);
SomfyRemote somfyRemote3(EMITTER_GPIO, REMOTE3, &rollingCodeStorage3);
SomfyRemote somfyRemote4(EMITTER_GPIO, REMOTE4, &rollingCodeStorage4);

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "xxxxxxxxxxxxx";

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "xxxxxxxxxxxxxx";
char pass[] = "xxxxxxxxxxxxxx";

// This function will be called every time Slider Widget
// in Blynk app writes values to the Virtual Pin 1

BLYNK_WRITE(V1) 
{
  int pinData = param.asInt();

  if (pinData == 1) // 1 is the MY button
  {
    const Command command = getSomfyCommand("1");
    somfyRemote1.sendCommand(command);
    }
  if (pinData == 2) // 2 is the down button
  {
    const Command command = getSomfyCommand("2");
    somfyRemote1.sendCommand(command);
    }
  if (pinData == 4) // 4 is the up button
  {
    const Command command = getSomfyCommand("4");
    somfyRemote1.sendCommand(command);
    }
  if (pinData == 8) // 8 is the program button
  {
    const Command command = getSomfyCommand("8");
    somfyRemote1.sendCommand(command);
    }
}
BLYNK_WRITE(V2)
{
  int pinData = param.asInt();

  if (pinData == 1) // 1 is the MY button
  {
    const Command command = getSomfyCommand("1");
    somfyRemote2.sendCommand(command);
    }
  if (pinData == 2) // 2 is the down button
  {
    const Command command = getSomfyCommand("2");
    somfyRemote2.sendCommand(command);
    }
  if (pinData == 4) // 4 is the up button
  {
    const Command command = getSomfyCommand("4");
    somfyRemote2.sendCommand(command);
    }
  if (pinData == 8) // 8 is the program button
  {
    const Command command = getSomfyCommand("8");
    somfyRemote2.sendCommand(command);
    }
}
BLYNK_WRITE(V3)
{
  int pinData = param.asInt();

  if (pinData == 1) // 1 is the MY button
  {
    const Command command = getSomfyCommand("1");
    somfyRemote3.sendCommand(command);
    }
  if (pinData == 2) // 2 is the down button
  {
    const Command command = getSomfyCommand("2");
    somfyRemote3.sendCommand(command);
    }
  if (pinData == 4) // 4 is the up button
  {
    const Command command = getSomfyCommand("4");
    somfyRemote3.sendCommand(command);
    }
  if (pinData == 8) // 8 is the program button
  {
    const Command command = getSomfyCommand("8");
    somfyRemote3.sendCommand(command);
    }
}
BLYNK_WRITE(V4)
{
  int pinData = param.asInt();

  if (pinData == 1) // 1 is the MY button
  {
    const Command command = getSomfyCommand("1");
    somfyRemote1.sendCommand(command);
    somfyRemote2.sendCommand(command);
    somfyRemote3.sendCommand(command);
    }
  if (pinData == 2) // 2 is the down button
  {
    const Command command = getSomfyCommand("2");
    somfyRemote1.sendCommand(command);
    somfyRemote2.sendCommand(command);
    somfyRemote3.sendCommand(command);    }
  if (pinData == 4) // 4 is the up button
  {
    const Command command = getSomfyCommand("4");
    somfyRemote1.sendCommand(command);
    somfyRemote2.sendCommand(command);
    somfyRemote3.sendCommand(command);    }
 }
void setup()
{
  // Debug console
  Serial.begin(115200);

  #if defined(ESP32)
  if (!EEPROM.begin(4)) {
    Serial.println("failed to initialise EEPROM");
    delay(1000);
  }
#elif defined(ESP8266)
  EEPROM.begin(4);
#endif

  Blynk.begin(auth, ssid, pass);
  // You can also specify server:
  //Blynk.begin(auth, ssid, pass, "blynk-cloud.com", 80);
  //Blynk.begin(auth, ssid, pass, IPAddress(192,168,1,100), 8080);
}

void loop()
{

  Blynk.run();
}`

closed 2021-04-13 by Legion2