Legion2/Somfy_Remote_Lib #12: Loss connection on restart

santinet28, 2021-06-17:
Hello,

I have managed to connect correctly with the motor, but when cutting the power and restarting it loses connection with the motor.
  I have looked at the debug and the rollingcode works fine, but still the engine is not responding.

I have also tried to load the CC1101 example and once synchronized then load the remotySomfy example. In this way the rollingcode also works well and the motor responds, the error turns off and on again.

Thank you

santinet28, 2021-06-17:
[image]

Legion2, 2021-06-17:
Hi,
so it works with the CC1101 example also after restarts, but your own code does not work after restarts? Can you provide the code which causes the problem?

santinet28, 2021-06-18:
gracias por responder

con el ejemplo cc1101 funciona perfecto después del reinicio y de apagar y encender.

El skecht que no funciona después de apagar és  el del ejemplo remoteSomfy

`
#include <EEPROM.h>
#include <EEPROMRollingCodeStorage.h>
#include <SomfyRemote.h>

#define EMITTER_GPIO 4  // I've changed it
#define EEPROM_ADDRESS 1 // and this 
#define REMOTE 0x5184c8

EEPROMRollingCodeStorage rollingCodeStorage(EEPROM_ADDRESS);
SomfyRemote somfyRemote(EMITTER_GPIO, REMOTE, &rollingCodeStorage);

void setup() {
	Serial.begin(115200);

	somfyRemote.setup();

#if defined(ESP32)
	if (!EEPROM.begin(4)) {
		Serial.println("failed to initialise EEPROM");
		delay(1000);
	}
#elif defined(ESP8266)
	EEPROM.begin(4);
#endif
}

void loop() {
	if (Serial.available() > 0) {
		const String string = Serial.readStringUntil('\n');
		const Command command = getSomfyCommand(string);
		somfyRemote.sendCommand(command);
#ifdef DEBUG
		Serial.println("finished sending");
#endif
	}
}

`

Now I have copied the setup lines of the example cc1101 and my skecht works correctly.
`
	ELECHOUSE_cc1101.Init();
	ELECHOUSE_cc1101.setMHZ(CC1101_FREQUENCY);
	ELECHOUSE_cc1101.SetTx();
`

The problem is resolved 

Thanks

closed 2021-06-18 by Legion2