Legion2/Somfy_Remote_Lib #17: How to support Open/Close button

rkotulan, 2022-01-22:
Hello,

My remote Telis 4 Modulis RTS5 Channel Pure Hand-Held Remote 1810765 has Open and Close button.
I can send prog, Up (one step), Down (one step), My command. 

But I do not know how to simulate Open (roll blind up) and Close (roll blind down).

Any ideas?

Legion2, 2022-01-22:
With my remote the Up and Down commands, open/close the blinds. Maybe you need to send to commands to repoduce this behavior with your remote?

rkotulan, 2022-01-22:
I can send commands repeatedly. But my remote control does it differently.

Legion2, 2022-01-22:
Did you try other commands?

rkotulan, 2022-01-22:
Yes, 0x1 to 0xF :)

Legion2, 2022-01-23:
Without more information about the protocol your remote is using, I can't help you fixing the problem.

rkotulan, 2022-01-23:
I understand, could you suggest some (easy) way (hw + sw) how to read data from my remote and decode it?

Legion2, 2022-01-23:
I have all the information about the protocol from [this](https://pushstack.wordpress.com/somfy-rts-protocol/) website and https://github.com/Nickduino/Somfy_Remote.

rkotulan, 2022-01-23:
Thanks, I will be back if I make some progress.

Legion2, 2022-01-23:
maybe #16 is related to this,

rkotulan, 2022-01-23:
Do you know what seed is in the frame?
_It looks like the seed isn't random but actually the button code: 0x88 DOWN, 0x85 STOP, 0x86 UP._

Legion2, 2022-01-23:
No I don't understand that part

vaguiner7, 2022-01-24:
Hello,

using the tool [rtl_433](https://github.com/merbanan/rtl_433) and the [dongle](https://pt.aliexpress.com/item/1005003259432535.html?gatewayAdapt=glo2bra&spm=a2g0o.9042311.0.0.4a55b90aOfDimZ) I could see the result below that explain: _It looks like the seed isn't random but actually the button code: 0x88 DOWN, 0x85 STOP, 0x86 UP._
[image]

Hex. codes of my remote control.

UP - f0f0ff34cb4b2ccb334ad2b4d2d532d353
STOP - f0f0ff34cccb2d4b32cb53355354b352d28
DOWN - f0f0ff34b54b534b4ccd55335552b554d48

rkotulan, 2022-01-24:
Thanks, what kind of remote do you have.
I tried these codes yesterday but it did not work for me.

I am going to order the dongle. :)

vaguiner7, 2022-01-24:
My remote is [SCE-R1 Gunmetal](https://www.tube-motors.com/worldwide/Item/4/165).

Yes, didn't work for me too, I think that is related to the buildFrame, the frame is different from the regular  Somfy products.

rkotulan, 2022-01-24:
You were able to decode your hex codes?

vaguiner7, 2022-01-24:
No, I used the tool rtl_433 to see my codes.

rkotulan, 2022-01-25:
@vaguiner7 The dongle came today. What command do you use for rtl_433?

vaguiner7, 2022-01-25:
Great,

I just used filtering the protocol somfy_rts: `rtl_433 -R 167`.

[image]

I really recommend the tool Universal Radio Hacker ([urh](https://github.com/jopohl/urh)) that you can use with the same dongle.

rkotulan, 2022-01-25:
@vaguiner7 Thank you so much.

Finally I was able to make it work. 

My problem was that method sendCC1101Command(Command command, int repeat = 4)  has default repeat = 4. I this case the motor make just one step. When I send just one command then the blinds come up :)

Legion2, 2022-01-25:
@rkotulan would you like to add some documenation?

rkotulan, 2022-01-26:
@Legion2 sure I will write something.

Leviathan09, 2022-02-28:
> Finally I was able to make it work.
> 
> My problem was that method sendCC1101Command(Command command, int repeat = 4) has default repeat = 4. I this case the motor make just one step. When I send just one command then the blinds go up :)

@rkotulan Could you maybe post your complete solution in here?
I have a curtain which is using the same remote as you have and i'm trying to figure out how to controll that with my WemosD1.
And i think i have to solve a similar problem as you, because with the UP/DOWN command i can't fully OPEN/CLOSE the curtain.

rkotulan, 2022-02-28:
@Leviathan09 Here is my code:
```c
#include "esphome.h"
#include <ELECHOUSE_CC1101_SRC_DRV.h>
#include <NVSRollingCodeStorage.h>
#include <SomfyRemote.h>

#define EMITTER_GPIO 2

#define CC1101_FREQUENCY 433.42

#define COVER_OPEN 1.0f
#define COVER_CLOSED 0.0f

#define TILT_OPEN 1.0f
#define TILT_CLOSED 0.0f

class SomfyESPCover : public Cover {
private:
  SomfyRemote *remote;
  NVSRollingCodeStorage *storage;

public:
  SomfyESPCover(const char *name, const char *key, uint32_t remoteCode)
      : Cover() {
    storage = new NVSRollingCodeStorage(name, key);
    remote = new SomfyRemote(EMITTER_GPIO, remoteCode, storage);
  }

  CoverTraits get_traits() override {
    auto traits = CoverTraits();
    traits.set_is_assumed_state(true);
    traits.set_supports_position(false);
    traits.set_supports_tilt(true);
    return traits;
  }

  void sendCC1101Command(Command command, int repeat = 1) {
    ELECHOUSE_cc1101.SetTx();
    remote->sendCommand(command, repeat);
    ELECHOUSE_cc1101.setSidle();
  }

  void control(const CoverCall &call) override {
    if (call.get_position().has_value()) {
      float pos = *call.get_position();

      if (pos == COVER_OPEN) {
        ESP_LOGI("somfy", "OPEN");
        sendCC1101Command(Command::Up);
      } else if (pos == COVER_CLOSED) {
        ESP_LOGI("somfy", "CLOSE");
        sendCC1101Command(Command::Down);
      } else {
        ESP_LOGI("somfy", "WAT");
      }

      this->position = pos;
      this->publish_state();
    }

    if (call.get_stop()) {
      ESP_LOGI("somfy", "STOP");
      sendCC1101Command(Command::My);
    }

    if(call.get_tilt().has_value()) {
      float tilt = *call.get_tilt();      
      ESP_LOGI("somfy", "tilt: %.1f",tilt );
      if(tilt == TILT_OPEN) {
        sendCC1101Command(Command::Up, 4);
      } else {
        sendCC1101Command(Command::Down, 4);
      }
    }
  }

  void program() {
    ESP_LOGI("somfy", "PROG");
    sendCC1101Command(Command::Prog);
  }
};

class SomfyESPRemote : public Component {
public:
  std::vector<esphome::cover::Cover *> covers;

  void setup() override {
    // need to set GPIO PIN 4 as OUTPUT, otherwise no commands will be sent
    pinMode(EMITTER_GPIO, OUTPUT);
    digitalWrite(EMITTER_GPIO, LOW);

    ELECHOUSE_cc1101.Init();
    ELECHOUSE_cc1101.setMHZ(CC1101_FREQUENCY);
  }

  void add_cover(const char *name, const char *key, uint32_t remoteCode) {
    auto cover = new SomfyESPCover(name, key, remoteCode);
    covers.push_back(cover);
  }
};
```
The key point is to change sendCC1101Command to send command just once. When you send Up = Open, Down = Close just once the cover will Open / Close.

I enabled also tilt support. If you do not need you can disable it traits.set_supports_tilt(false);

rkotulan, 2022-02-28:
@Legion2 
Documentation update
For remote control Telis 4 Modulis RTS5 I recommend sending the command Up or Down just once. 
When the UP command is sent once, the blinds go up (Open). 
```c
sendCommand(Command::Up, 1);
```

When the DOWN command is sent once, the blinds go down (Close). 
```c
sendCommand(Command::Down, 1);
```

If you want to tilt the blinds, send the Up / Down command four times.
```c
sendCommand(Command::Up, 4);
```
or
```c
sendCommand(Command::Down, 4);
```

Legion2, 2022-02-28:
@rkotulan can you create a PR, else I can update the readme.

Leviathan09, 2022-03-01:
> The key point is to change sendCC1101Command to send command just once. When you send Up = Open, Down = Close just once the cover will Open / Close.
> 
> I enabled also tilt support. If you do not need you can disable it traits.set_supports_tilt(false);
> 
> If you need more details let me know.

@rkotulan Thanks, really appreciate this.
Will check the code and try to understand it as i'm not that good at coding such stuff.

Leviathan09, 2022-03-01:
> @Leviathan09 I use ESPHome + HomeAssistatnt.
> 
> How do you imagine you will control the curtain? With button connected to wemos? It should not be a problem to rewrite the code for Arduino.

Okay, then i will try to figure out how ESPHome works or will check if a can adjust it for ArduinoIDE (have the example from MakerMeiks Somfy_Remote)
My plan is to use one of my dashboards from iobroker on a small tablet or my phone...and maybe also my Amazon Echo at least for full Open/Close

rkotulan, 2022-03-01:
@Leviathan09  this might help https://github.com/DrozmotiX/ioBroker.esphome

Leviathan09, 2022-03-02:
@rkotulan i tried it with ESPHome and HomeAssistant and was able to perform a compile process but it failed and i don't know why. As already mentiond i'm not really good and such "programming" stuff.
Maybe you can help me.

[logs_esp-curtain_compile-1.txt](https://github.com/Legion2/Somfy_Remote_Lib/files/8168172/logs_esp-curtain_compile-1.txt)

rkotulan, 2022-03-02:
@Leviathan09 could you post here file my_custom_component.h ?

Leviathan09, 2022-03-02:
> @Leviathan09 and how you YAML for ESPHome looks like?

esphome:
  name: esp-curtain
  includes:
  - my_custom_component.h

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "653737a9ae00761ac8dcb44766154ed2"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Curtain Fallback Hotspot"
    password: "78mTcz5ZpI5E"

captive_portal:

Leviathan09, 2022-03-02:
> @Leviathan09 and how you YAML for ESPHome looks like?

[yaml.txt](https://github.com/Legion2/Somfy_Remote_Lib/files/8168471/yaml.txt)

rkotulan, 2022-03-02:
@Leviathan09  One thing, I have different board then you. I have similar to this one https://www.aliexpress.com/item/1005003818247483.html

Leviathan09, 2022-03-02:
hm, okay
i think the board shouldn't be a problem

But as i don't really understand your stuff from above i think i have to wait for the weekend to take enough time
But thank you for your help so far

rkotulan, 2022-03-02:
@Leviathan09 For build board is not problem but this line
```c
#define EMITTER_GPIO 2
```
could depend on the board, but I am not sure.

rkotulan, 2022-03-06:
@Legion2 please update documentation

filzek, 2022-12-01:
> Great,
> 
> I just used filtering the protocol somfy_rts: `rtl_433 -R 167`.
> 
> [image]
> 
> I really recommend the tool Universal Radio Hacker ([urh](https://github.com/jopohl/urh)) that you can use with the same dongle. [image]

@vaguiner7  did you make the Library to work with TUBE? As we try to use it the TUBE seens to have a 7 Digits Hex ID, so, whenever we set it 6 or 7 digits it doest work.

Can you give me a help to try to solve this?

Using a RFLINK to sniff the RF communication it shows it as RTS and the Remote Control ID as 265822A 

Also if @Legion2  could take a look at this would be awesome!

commit dccd3c57 (2022-03-07): update documentation close #17