Legion2/Somfy_Remote_Lib #19: Multiple Remotes not working

Trabbi1999, 2022-03-15:
Hi,

I'm currently trying to realize a CC1101 Multiple Remote Setup via MQTT. Everything is working fine so far, the script does what is should do, BUT it fails at the most important point: sending out the command via the CC1101. The main problem is located in 

`void sendCC1101Command(Command command)`

The only working remote is somfyRemote1. When I send the value 100 via MQTT to the topic home/buero/rolladen the correct shutter is moving upwards. When I send 100 to the other topics defined in the top of the script, everything is working, despite the sending, all Serial.println are displayed correctly and I can confirm that that part is working. I used parts of your example "Multiple Remotes" and modified it for my use case, but its not working. 

I tested all shutters seperately with the CC1101 example script with the same Remote codes as defined in my multiple Remote script (I changed them to XXXX for github, thats not the error :P) and its working perfectly fine. 

All remotes are configured the same, but only one is working. Do you have any idea why?

Here is the script (unfortunately, the github script function does not work) :

`#include <PubSubClient.h>
#include <WiFi.h>
#include <EEPROM.h>
#include <EEPROMRollingCodeStorage.h>
#include <ELECHOUSE_CC1101_SRC_DRV.h>
#include <SomfyRemote.h>
#include <ArduinoOTA.h>
#define DEBUG
#define EMITTER_GPIO 2
#define REMOTE1 0xXXXXX
#define REMOTE2 0xXXXXX
#define REMOTE3 0xXXXXX
#define REMOTE4 0xXXXXX
int use_remote;

#define wifi_ssid "XXXX"
#define wifi_password "XXXX"
#define mqtt_server "XXXXXXXXX"
#define mqtt_user "XXXXXXXXXXX"
#define mqtt_password "XXXXXXXXXXXX"
#define ESPHostname "ESP32_MQTT_Rolladen"
String clientId = "ESP32-Rolladen-";

#define CC1101_FREQUENCY `433.42`

//Topics zuordnen
#define inTopic1 "home/buero/rolladen"
#define inTopic2 "home/kueche/rolladen"
#define inTopic3 "home/wohnzimmer/rolladen/links"
#define inTopic4 "home/wohnzimmer/rolladen/rechts"

WiFiClient espClient;
PubSubClient client(espClient);

EEPROMRollingCodeStorage rollingCodeStorage1(0);
EEPROMRollingCodeStorage rollingCodeStorage2(2);
EEPROMRollingCodeStorage rollingCodeStorage3(4);
EEPROMRollingCodeStorage rollingCodeStorage4(6);
//Erstellung Remotes
SomfyRemote somfyRemote1(EMITTER_GPIO, REMOTE1, &rollingCodeStorage1);
SomfyRemote somfyRemote2(EMITTER_GPIO, REMOTE2, &rollingCodeStorage2);
SomfyRemote somfyRemote3(EMITTER_GPIO, REMOTE3, &rollingCodeStorage3);
SomfyRemote somfyRemote4(EMITTER_GPIO, REMOTE4, &rollingCodeStorage4);

void setup() {

  Serial.begin(115200);
  Serial.println("Hello World");
  setup_wifi();
  ArduinoOTA.setHostname(ESPHostname);
  ArduinoOTA.begin();

  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);

  somfyRemote1.setup();
[... 253 lines cut ...]
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Create a random client ID
    clientId += String(random(0xffff), HEX);
    // Attempt to connect
    if (client.connect(clientId.c_str(), mqtt_user, mqtt_password)) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      //client.publish(/*outTopic, */ESPHostname);
      // ... and resubscribe
      client.subscribe(inTopic1);
      client.subscribe(inTopic2);
      client.subscribe(inTopic3);
      client.subscribe(inTopic4);
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}`

Thanks! :)

Legion2, 2022-03-19:
During separate testing, did you also use the correct rollingcodestorage for the different remotes?

PS: I fixed the GitHub code block syntax in your comment by using three ` and a new line

Trabbi1999, 2022-03-20:
OMG Danke :D that was the problem. Everything is working now. But, antoher question/issue:

I took over the part with the EEPROM Adresses from your multiple remote skript, as you can see above (0,2,4,6). Unfortuantely, only 0 and 2 are working correctly. With higher values I can program the shutters, but i cannot control them. It's working now with following eepromaddresses:

somfyremote1: 0
somfyremote2: 2 
somfyremote3: 0
somfyremote4: 0

As I said, its working with that setup, but will the multiple use of the address 0 cause issues? or shouldn't it be a problem?

Danke :)

Legion2, 2022-03-20:
I you use the same address for multiple stores, they override each other and cause multiple issues. You can try to reset the EEPROM at that addresses to `0` and reprogram the shutters.

closed 2022-10-28 by Legion2