Legion2/Somfy_Remote_Lib #21: unabel to control covers

ageurtse, 2022-09-05:
I have a problem, but don't know where to look. 
I rebuild the schematic from this site [https://www.die-welt.net/2021/06/controlling-somfy-roller-shutters-using-an-esp32-and-esphome/](https://www.die-welt.net/2021/06/controlling-somfy-roller-shutters-using-an-esp32-and-esphome/)

flashed my ESP32, but the up/down and stop commands won't work. 
When learning the remote the cover goes shortly up an down. 
but that is it. 

i make use of NVSRollingCodeStorage.

Legion2, 2022-09-05:
What type of covers do you have and what type of remote did you used before? Have a look at https://github.com/Legion2/Somfy_Remote_Lib/issues/17#issuecomment-1054628543, you  may also have to adjust the `repeat` parameter for your type of covers to work.

ageurtse, 2022-09-06:
I have blinds witw a somfy rts motor.
The origanaly remote is an old telis 4

ageurtse, 2022-09-08:
tryed the things from post 17, and changed the repeat time's but that didn't solved it. 
any more clue's

Legion2, 2022-09-08:
I think your covers and remote use different commands or variations of the commands. Maybe you find some more information about their protocol in the internet (e.g. https://github.com/merbanan/rtl_433), but I can't help you debug your hardware.

adam-thomas-privitar, 2022-11-29:
Hi. I have this exact same issue. Prog works, the blinds jump when I run PROG. But after, nothing happens -- no response on the up or down.

I am purchasing an RTL-SDR stick to check and compare the signals myself. I will report back if I have luck finding this issue.

Maybe its the blinds we have? I have "Electra" blinds which integrate with somfy remote. Is it the same for you? I can't find the model on the damn things.

ageurtse, 2022-11-29:
I abbonded ghis and installed an rflink gateway, which worked great.

adamscybot, 2022-11-29:
@ageurtse Interesting. Does that thing support somfy out of the box? If so, perhaps I can work backwards from its software to how it sends the commands?

Or do you think its a hardware issue?

ageurtse, 2022-11-29:
Yes it works out of the box but its based on a arduino mega. 

All the info is here [https://www.rflink.nl/](https://www.rflink.nl/)

adamscybot, 2022-11-29:
Thanks. Apparently the source is now closed https://www.letscontrolit.com/forum/viewtopic.php?t=8323 so my plan wont work. Though I did find https://github.com/filipmaelbrancke/ha-rflink-rts.

Whatever its doing inside RFLink, this project is not. Hopefully I can shine light on it with the dongle.

adamscybot, 2022-12-04:
I have purchased an RTL SDR and decoded the stream. I've dumped some info here https://docs.google.com/spreadsheets/d/1AV9f8sxG4KNrnWUGNh_Anl8WvctlWDfXvrI9FVzuEMo/edit?usp=sharing.

The somfy_decode program doesn't pick it up, it thinks it enters the preamble and immediately exits the preamble on the next data. I added debugging to capture the above.

So probably, unless my capture is bad quality somehow, the protocol is indeed different. Hmmm.

adamscybot, 2022-12-04:
More data from rtl_433:

**Pressing UP**

```
Detected OOK package    2022-12-04 18:42:01
Analyzing pulses...
Total count:   94,  width: 251.32 ms            (62829 S)
Pulse width distribution:
 [ 0] count:   14,  width: 2508 us [2504;2520]  ( 627 S)
 [ 1] count:    2,  width: 4840 us [4836;4844]  (1210 S)
 [ 2] count:   34,  width: 1308 us [1296;1332]  ( 327 S)
 [ 3] count:   44,  width:  668 us [652;688]    ( 167 S)
Gap width distribution:
 [ 0] count:   14,  width: 2500 us [2492;2512]  ( 625 S)
 [ 1] count:   34,  width: 1260 us [1248;1276]  ( 315 S)
 [ 2] count:   44,  width:  616 us [592;632]    ( 154 S)
 [ 3] count:    1,  width: 27468 us [27468;27468]       (6867 S)
Pulse period distribution:
 [ 0] count:   16,  width: 5148 us [5004;6100]  (1287 S)
 [ 1] count:   20,  width: 2568 us [2564;2580]  ( 642 S)
 [ 2] count:   24,  width: 1928 us [1912;1940]  ( 482 S)
 [ 3] count:   32,  width: 1284 us [1272;1296]  ( 321 S)
 [ 4] count:    1,  width: 28768 us [28768;28768]       (7192 S)
Level estimates [high, low]:  15915,   4450
RSSI: -0.1 dB SNR: 5.5 dB Noise: -5.7 dB
Frequency offsets [F1, F2]:   19767,      0     (+75.4 kHz, +0.0 kHz)
Guessing modulation: No clue...

```

**Pressing down**

```
Detected OOK package    2022-12-04 18:42:04
Analyzing pulses...
Total count:  251,  width: 661.45 ms            (165363 S)
Pulse width distribution:
 [ 0] count:    1,  width: 10428 us [10428;10428]       (2607 S)
 [ 1] count:   30,  width: 2512 us [2500;2532]  ( 628 S)
 [ 2] count:    5,  width: 4840 us [4832;4856]  (1210 S)
 [ 3] count:   65,  width: 1308 us [1296;1328]  ( 327 S)
 [ 4] count:  150,  width:  668 us [652;700]    ( 167 S)
Gap width distribution:
 [ 0] count:    1,  width: 7104 us [7104;7104]  (1776 S)
 [ 1] count:   30,  width: 2496 us [2472;2508]  ( 624 S)
 [ 2] count:   65,  width: 1256 us [1228;1272]  ( 314 S)
 [ 3] count:  150,  width:  612 us [592;632]    ( 153 S)
 [ 4] count:    4,  width: 27428 us [27400;27472]       (6857 S)
Pulse period distribution:
 [ 0] count:    1,  width: 17532 us [17532;17532]       (4383 S)
 [ 1] count:   35,  width: 5164 us [5000;6096]  (1291 S)
 [ 2] count:   20,  width: 2568 us [2560;2580]  ( 642 S)
 [ 3] count:   85,  width: 1924 us [1916;1940]  ( 481 S)
 [ 4] count:  105,  width: 1284 us [1272;1296]  ( 321 S)
 [ 5] count:    4,  width: 28096 us [28068;28136]       (7024 S)
Level estimates [high, low]:  16067,   4705
RSSI: -0.1 dB SNR: 5.3 dB Noise: -5.4 dB
Frequency offsets [F1, F2]:   20154,      0     (+76.9 kHz, +0.0 kHz)
Guessing modulation: No clue...

```

adamscybot, 2022-12-05:
Ah so I was using an old version of rtl_433 by mistake. With a newly compiled version, I can see the frames picked up from the original remote.

Ive masked out my original remote ID below.

**Down**:

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:32:16
model     : Somfy-RTS    id        : XXXXXX
Control   : Down (4)     Counter   : 2306          Retransmission: 1         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:32:16
model     : Somfy-RTS    id        : XXXXXX
Control   : Down (4)     Counter   : 2306          Retransmission: 1         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:32:17
model     : Somfy-RTS    id        : XXXXXX
Control   : Down (4)     Counter   : 2307          Retransmission: 0         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:32:17
model     : Somfy-RTS    id        : XXXXXX
Control   : Down (4)     Counter   : 2307          Retransmission: 1         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:32:17
model     : Somfy-RTS    id        : XXXXXX
Control   : Down (4)     Counter   : 2307          Retransmission: 1         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:32:17
model     : Somfy-RTS    id        : XXXXXX
Control   : Down (4)     Counter   : 2307          Retransmission: 1         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:32:17
model     : Somfy-RTS    id        : XXXXXX
Control   : Down (4)     Counter   : 2307          Retransmission: 1         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:32:20
model     : Somfy-RTS    id        : XXXXXX
Control   : Down (4)     Counter   : 2309          Retransmission: 1         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:32:20
model     : Somfy-RTS    id        : XXXXXX
Control   : Down (4)     Counter   : 2309          Retransmission: 1         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:32:21
model     : Somfy-RTS    id        : XXXXXX
Control   : Down (4)     Counter   : 2310          Retransmission: 1         Integrity : CHECKSUM
```

**Up**:

```
time      : 2022-12-05 00:34:18
model     : Somfy-RTS    id        : XXXXXX
Control   : Up (2)       Counter   : 2313          Retransmission: 1         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:34:18
model     : Somfy-RTS    id        : XXXXXX
Control   : Up (2)       Counter   : 2313          Retransmission: 1         Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:34:18
model     : Somfy-RTS    id        : XXXXXX
Control   : Up (2)       Counter   : 2313          Retransmission: 1         Integrity : CHECKSUM
[... cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-12-05 00:34:18
model     : Somfy-RTS    id        : XXXXXX
Control   : Up (2)       Counter   : 2313          Retransmission: 1         Integrity : CHECKSUM
```

So this confirms my problem is not caused by the blinds working on different commands.

It is of note I have the same CC1101 as this guy: https://github.com/Legion2/Somfy_Remote_Lib/issues/22.

Now to check how the signal appears when sent from my fake remote.

adamscybot, 2022-12-05:
I have discovered the reason RTL_433 does not detect up/down frames from my fake remote is that it fails this check in its decoder:

```
    if (bitbuffer_search(bitbuffer, decode_row, 0, preamble_pattern, preamble_pattern_bit_length) != 0) {
        printf("Abort early 2\n");
        return DECODE_ABORT_EARLY;
    }
```

Where with the real remote it passes, and PROG passes on the fake one too.

adamscybot, 2022-12-05:
Aaaand BINGO!

I have it working. The problems were as follows:

* The `key` that I used for the NVS rolling storage was long and had spaces, probably too long.
* At the same time, I changed the name of the storage to abandon the old area of the NVS storage.
* Reprogrammed.
* It now all works.

I would hazard a guess people who have this problem have the same issue.

adamscybot, 2022-12-11:
I've gone ahead and raised a PR to add a bit more info

Legion2, 2023-04-03:
I merged the documentation update and now closing this issue as resolved.

closed 2023-04-03 by Legion2

evgeni, 2025-02-22:
FWIW, When moving the code to an "external component" (https://github.com/evgeni/esphome-configs/issues/113) I also implemented limits, so people should get proper errors when they try to use namespace or storage names longer than 15 chars :)