Legion2/Somfy_Remote_Lib #22: Not able to PROG a new remote - ESP8266 and E07-M1101d

sl4m01, 2022-11-04:
Dear all,
it seems like the TI CC1101 are very hard to get currently.
I ordered one at Amaz*n but got a E07-M1101D.
I googled it and it seems its the Version 2 or the CC1101 (?).
However the pinning looks different, its not 2 row by 5 pins, its now 2 rows by 4 pins:
[image]

I checked the specification and found this: 
[image]
[image]
So the pins are the same like on the TI CC1101, except the additional VCC and GND.
So i wired it like described in the SmartRC-CC1101-Driver-Lib
[image]
But not to a WEMOS D1 mini but to an ESP8266 NodeMCU, hoping that i can just use the same D# Pins:
[image]
[image]

Now i ran the CC1101 example from the Somfy_Remote_Lib and set #define CC1101_FREQUENCY 433.42.
Set my Somfy RTS control to Prog (blinds go up down) and typed Prog in the Serial Monitor (set to new line and 115200 Baud) but nothing happens (also no answer in the serial monitor). After ~2 minutes the blinds go again up and down (i think this means the prog-duration expired).
Than i  tried the cc1101_Receive_minimal example of the SmartRC-CC1101-Driver-Lib and at least i am getting a 
10:27:30.655 -> Connection OK
10:27:30.655 -> Rx Mode
there. But also no response if i press a button on my Somfy remote. 

Now i dont know if the E07-M1101D is not supported, or my wiring is wrong, or i typed something wrong in the serial monitor :(
Can someone try to help?
Thank you alot!

Legion2, 2022-11-05:
Can you ask in the SmartRC-CC1101-Driver-Lib project if the E07-M1101D is supported? Then we can narrow down the issue.

sl4m01, 2022-11-06:
Thank you for your answer. 
I just wanted to do this but then I saw that on the git was no issue answer since 2 years. Don't expect to get an answer there. Any other way to test what the problem is? If not I would try to get a TI Cc1101

sl4m01, 2022-11-08:
I found this project 
https://github.com/ESPboy-edu/ESPboy_Sub1GHzInspector 
and they are using the E07-M1101D in combination with the SmartRC-CC1101-Driver-Lib

for blue CC1101 version

1 >> GND
2 >> VCC
3 >> D4, GPIO2
4 >> D8, GPIO15, SS
5 >> D5, GPIO14, SCK
6 >> D7, GPIO13, MOSI
7 >> D6, GPIO12, MISO
8 >> RX, GPIO3, RX

Found the following in ESPboy_Sub1GHzInspector.ino:

`#include <ELECHOUSE_CC1101_SRC_DRV.h>
#include <RCSwitch.h>
#include <ESP_EEPROM.h>

#define CC1101riceivePin    3
#define CC1101sendPin       2
#define CC1101chipSelectPin D8`

So my assumption is that it is supported

Legion2, 2022-11-08:
I looked at the [ESPboy_Sub1GHzInspector](https://github.com/ESPboy-edu/ESPboy_Sub1GHzInspector) project and cannot find a reference to E07-M1101D.

sl4m01, 2022-11-09:
Thank you for checking the project. The upper blue CC1101 is exactly the module I am using. Also they are using it in their build up (at least I can see 2x4 pins and not 2x5)
[image]
I checked Google and eBay again and I am not able to find an available TI cc1101 :( nowhere on stock anymore

Legion2, 2022-11-10:
As you can see the 2x5 has duplicate VCC and GND pins, I think they are removed with the 2x4, but they also had to reorder the other pins. So we need to find the pin out diagram for the 2x4 to wire it correctly, then it should work with the SmartRC-CC1101-Driver-Lib.

sl4m01, 2022-11-11:
I found this specification for the E07-M1101D
[image]
[image]

And currently i am using this wiring:
[image]

However, following the wiring they are using here https://github.com/ESPboy-edu/ESPboy_Sub1GHzInspector it should be like this:
[image]
However there are some #defines in their code:
`#include <ELECHOUSE_CC1101_SRC_DRV.h>
#include <RCSwitch.h>
#include <ESP_EEPROM.h>

#define CC1101riceivePin 3
#define CC1101sendPin 2
#define CC1101chipSelectPin D8`

I can try both setups but i am little bit lost here :(

Legion2, 2022-11-22:
Maybe your receiver does not understand the send commands, because it uses some other protocol. Have you tried to use a different parameter as second argument to the `sendCommand` function?

sl4m01, 2022-11-23:
Hello, no i didnt try. Do you mean something like sendCommand(Command::Prog, 2); ?

Legion2, 2022-11-23:
Yes or use parameter 4, see readme for more details.

adamscybot, 2022-12-05:
I have this same CC1101 and experiencing same issue. Unsure if its related to this hardware or not, but see here https://github.com/Legion2/Somfy_Remote_Lib/issues/21

MrBLJ, 2023-03-17:
Same hardware here, I got it _kinda_ working with a D1 mini and nodemcu, both esp8266 based. I say kinda, because although I can transmit commands, the signal is very weak and not recognized by rtl_433 for example. Using airspy, I can see that something is transmitted, near the frequency of the original remote, but with a slight offset :

[image]

I'm using the [CC1101 sketch](https://github.com/Legion2/Somfy_Remote_Lib/blob/main/examples/CC1101/CC1101.ino).

Has anyone encountered the same issue ?

adam-thomas-privitar, 2023-03-17:
That offset is likely not enough to cause an issue but the signal power is likely an issue. What antenna are you using with your CC1101? Double check it is properly connected. possibly also check the CC1101 is getting the right voltage.

It could also be faulty. There's lots of crappy PCBs out there.

MrBLJ, 2023-03-20:
> That offset is likely not enough to cause an issue (mine is similar) but the signal power could be an issue. What antenna are you using with your CC1101? Double check it is properly connected. possibly also check the CC1101 is getting the right voltage with a multimeter.
> 
> The CC1101 could also be faulty. There are lots of crappy PCBs out there.
> 
> Another totally different and very possible scenario (maybe even more possible) is it is not wired up right. It's odd its just a single line with no pattern, even if power is reduced. Many ESPs have differing pins to each other. Would be useful to share pictures/diagram of the wiring and exact model number of the ESP and CC1101. You should also thoroughly check for shorts between the CC1101 pins.

Thanks for your answer.

I'm using the same board and antenna as displayed above 
[image]

I have two of them, wired to ESP8266 :
- D1 mini with dupont connectors.
- NodeMCU directly soldered to the pins. Tested for shorts and continuity.

All boards have been cleaned, using proper PCB cleaning solvents, and then dried with compressed air.

NodeMCU :

[image]
[image]

(My soldering is not the prettiest) 

D1 mini :
[image]
[image]
[image]

I'm still getting some signal when I send a command, as seen in airspy :
[image]
[image]

But nothing close to what I'm getting with the remote in term of strength and amplitude (sorry if I misuse the terms).

adamscybot, 2023-09-01:
Hmmm.  Its certainly interesting, given I have the same CC1101 _and_ the NodeMCU as well. So pretty much same setup. Not to mention @MrBLJ soldering skills are a damn sight better than mine.

Still though I think you have to rule out software problems here. I also tried receiving my by rtl_433 and its _very very_ unreliable even on my actually-working setup. I would not use this as a trustworthy indicator.

The low signal strength could easily be the gain setting on your RTL? This is a common reason for this sort of thing.

When I say software problem -- what I mean is, do you blinds not respond to the PROG? If they do, probably you might actually be experiencing the blinds ignoring the signal due to the rolling code...

BasvanH, 2023-09-02:
After a very long night I have solved the issue. I got it working with an ESP8266 wemos D1 (clone).

If you have followed the [wiring of the SmartRC-CC1101-Driver-Lib guide](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/tree/master#wiring), then you have connected CC1101 to the ESP8266 like this:

CC1101E07-M1101d | ESP8266
-- | --
1 GND | GND
2 VCC | 3V3
3 GD00 | D1 GPIO5
4 CKN/CSN | D8 GPIO15
5 SCK | D5 GPIO14
6 MOSI | D7 GPIO13
7 MISO/GD01 | D6 GPIO12
8 GD02 | D2 GPIO4

In the example of the [CC1101.ino](https://github.com/Legion2/Somfy_Remote_Lib/blob/main/examples/CC1101/CC1101.ino) the EMITTER_GPIO is 2 (D4 on a ESP8266) which is wrong and should be set to 5, which is D1 connected to GD00. The very small signal we were seeing was probably because of some cross GPIO leaking within the ESP8266.

I now have a correct working CC1101E07-M1101d with this library.

I do recommend looking at the [new non GDO way](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/tree/master/examples/CC1101%20default%20examples) of using the SmartRC-CC1101-Driver-Lib library, this way you dont need to set the GDO pin anymore.

And secondly, there's a bug in [SomfyRemote.cpp](https://github.com/Legion2/Somfy_Remote_Lib/blob/main/src/SomfyRemote.cpp). 

https://github.com/Legion2/Somfy_Remote_Lib/blob/50a440a3038199d9d6b3b92471c13de4eb21318c/src/SomfyRemote.cpp#L15

The repeat is swapped with the rollingCode

https://github.com/Legion2/Somfy_Remote_Lib/blob/50a440a3038199d9d6b3b92471c13de4eb21318c/src/SomfyRemote.cpp#L18

I have a PR ready, will push it after this post.

amonwong, 2024-06-15:
I am literally stuck on this but with an Arduino Uno and E07-M1101d. I've wired it exactly like this and nothing happens. Can someone point me in the right direction?
[image]