Legion2/Somfy_Remote_Lib #24: Remote code With 7 Position (Tube RTS)

filzek, 2022-12-01:
Hi,

There is a problem to create/pair/use the library with remotes that use a 7 position ID besides 6 position.

The TUBE uses a RTS protocol but the remote identifier start with a 2 and follow 6 position as ID od the remote control.

Could you add an fix to be able.to use the remote address as 0x2123456 position on hex identifiers?

Thanks

filzek, 2022-12-01:
I did try everything but seens that is impossible to send correctly, can you check on this?

Legion2, 2022-12-04:
You mean the remote id is 26 bits for TUBE instead of the 24 bits of the normal RTS protocol? Currently the 24 bits are send in 3 * 8 bit frames. How do the 26 bits fit into the frames of the RTS protocol? Are they send as 4 * 8 bit frames with some padding? Where do you get the information with the 26 bits get from. Maybe its a 32 bits id?

filzek, 2022-12-05:
Not sure how the protocol is formed, as it is being recognized as SOMFY, identify correct all the commands,. but the ID identified is not the same or the protocol used is missing something. One interesting part it that TUBE is tuned at 433.92MHz. I

I just bought a SDL receiver to use with the RTL_433 to try to identify the signal and it also recognizes it as 433.92MHz and set it to SOMFY mode. 

But when trying to send the remote it doesnt work in somfy_remote_lib, even add incremental, so protocol is missing soemthing in TUBE mode.

filzek, 2022-12-09:
Maybe the time among sync, high low must be more precise? 

using a RTL-SDR analyzer it shows:

https://triq.org/pdv/#AAB0110701273009EC12F805000280FFFF806C8555+AAB03C0701273009EC12F805000280FFFF806C9191A3B4C3B3C4B4C3C4B4C4C4C3B3C4B4C3B3B4C4C3B4C4C4C3B4C4C4C3B3C4B3B4C4C3C4C4B4C4C3B4C4C655+AAB0410701273009EC12F805000280FFFF806C91919191919191A3B4C3B3C4B4C3C4B4C4C4C3B3C4B4C3B3B4C4C3B4C4C4C3B4C4C4C3B3C4B3B4C4C3C4C4B4C4C3B4C4C555

per you knowdledge maybe you could do a TUNE UP to we test in the timmings?

Legion2, 2022-12-10:
You can change the timings here in the code: https://github.com/Legion2/Somfy_Remote_Lib/blob/7b0c2f73401c623dba3ec3d004f547355af234f4/src/SomfyRemote.cpp#L76-L106

filzek, 2022-12-11:
I think I have found something interesting, usint the RTL_433 to decode, i have played to try to find the timmings and extra info for the remote control and I have found this:

we have add  seed to be printed to check if the seed/key is changed among comands or keys pressed as show:

a     /* clang-format off */
    data = data_make(
            "model",          "",               DATA_STRING, "Somfy-RTS",
            "id",             "",               DATA_FORMAT, "%06X", DATA_INT, address,
            "control",        "Control",        DATA_STRING, control_str,
            "counter",        "Counter",        DATA_INT,    counter,
            "retransmission", "Retransmission", DATA_INT,    is_retransmission,
            "mic",            "Integrity",      DATA_STRING, "CHECKSUM",
            "seed Decimal",   "seed",           DATA_INT,    seed,
            "seed Hex",       "",               DATA_FORMAT, "0x%0X", DATA_INT, seed,
            NULL);
    /* clang-format on */

and this is quite interesting to check as, for each key (UP, DOWN, STOP, PROG) it uses an exclusive key as per:
KEY // seed/key
UP // 0x96 hex
STOP // 0x95 hex
DOWN // 0x98 hex
PROG // 0x9C hex

So, it seens that the TUBE uses a fixed seed per key pressed.

Also the timmings are a little better adjusted than the SOMFY Trellis remote. 

We will try to reproduce (clone) the remote using this new parameters so this could be the trick to use the Somfy library correctly.

filzek, 2022-12-11:
Checking the SOMFY remote control its cycle the SEED by A0 to AF once every key is pressed.

In the TUBE it doesnt happen at all and use a FIXED as in the above post!

I will try to get more to know in 2 days when I wil lhave 2 more TUBE control

filzek, 2022-12-16:
@Legion2 I have tested the new updated code to work with the fixed code and it works fine, I will rework the code to propose an update to the lib and also example!

filzek, 2023-02-02:
@Legion2  we solve and the problem, that is the deal:

we have found all changes to use with SOMFY RTS TUBE

frame[0] = 0x95; // is the COMMAND FRAME  (in RTS is the KEY frame)
frame[1] = (unsigned char)(15)<< 4;  //is fixed always (in RTS was the CMD)

so, the CMD as off:
#define TUBE_SCE_R1_CMD_PROG 0x9C
#define TUBE_SCE_R1_CMD_STOP 0x95
#define TUBE_SCE_R1_CMD_UP 0x96
#define TUBE_SCE_R1_CMD_DOWN 0x98
#define TUBE_SCE_R1_CMD_UP_DOWN 0x9A

Repeate the FRAMES to use special Command:
to make MY working just sent the STOP 10 times
to make the PROG to work just send it 25 times!

so all is done!!!